gregdavill/kicadScripts #1: missing GetBoardEdgesBoundingBox attribute

clacktronics, 2018-04-03:
Hello, I am trying this with Kicad 4.07 and python 2.7 on Ubuntu and I get the following error

```bash
python plot_board.py ../board.kicad_pcb 
12:08:38: Debug: Skipping general section token 'links' 
12:08:38: Debug: Skipping general section token 'area' 
12:08:38: Debug: Skipping general section token 'drawings' 
12:08:38: Debug: Skipping general section token 'tracks' 
12:08:38: Debug: Skipping general section token 'zones' 
12:08:38: Debug: Skipping general section token 'symbol' 
Traceback (most recent call last):
  File "plot_board.py", line 367, in <module>
    bb = board.GetBoardEdgesBoundingBox()
  File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 4806, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, BOARD, name)
  File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 83, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'BOARD' object has no attribute 'GetBoardEdgesBoundingBox'
```
I can confirm I do not have `GetBoardEdgesBoundingBox` in `pcbnew.py` the line number is significantly lower than http://docs.kicad-pcb.org/doxygen-python/pcbnew_8py_source.html#l19204. Curious if you have any ideas!

gregdavill, 2018-04-04:
I'm running KiCad 5 rc2. They've obviously added the `GetBoardEdgesBoundingBox` function. A quick fix is to change this to `GetBoardBoundingBox`.

Using BoardEdges was an easy way to better crop the final image. A better method will be to calculate a bounding box based on the EdgeCuts layer once it's in SVG form.

gregdavill, 2019-04-21:
I'm going to assume that people have upgraded to KiCad 5.1 at this point, and close this issue.

closed 2019-04-21 by gregdavill