gregdavill/kicadScripts #2: Getting the plot_board script up and running on MacOS.

MichMich, 2019-03-24:
First of all, thanks for your awesome work and inspiring tweets. It really pushes me to get better at electronics.

So, now for the issue: I'm trying to get the code up and running on MacOS (10.14.3). But unfortunately I'm getting some unexpected results which might have something to do with masking.

Initially I just tried running `python plot_board.py path/to/kicad_project.kicad_pcb` but this gave the following error on MacOS:

```
Traceback (most recent call last):
  File "plot_board.py", line 9, in <module>
    import pcbnew
ImportError: No module named pcbnew
```
With the help of the KiCad forum I figured out I needed to make some modifications to the plot_board script:
* Add a shebang line to the top: `#!/Applications/Kicad/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python`
* Use the following code after the shebang line:
```
import sys
sys.path.insert(
    0,
    "/Applications/Kicad/kicad.app/Contents/Frameworks/python/site-packages/")
```
* Modify the executalbe path of inkscape on line 311 to: `/Applications/Inkscape.app/Contents/Resources/bin/inkscape` 
* Make the script executable: `chmod +x plot_board.py`

After these changes, I'm able to run the script using the following command: `./plot_board.py path/to/kicad_project.kicad_pcb`

The script still gives a bunch of errors: 
```
** (inkscape-bin:1123): WARNING **: attribute 'mask' given as CSS
** (inkscape-bin:1123): WARNING **: attribute 'mask' given as CSS
** (inkscape-bin:1123): WARNING **: attribute 'mask' given as CSS
** (inkscape-bin:1123): WARNING **: attribute 'mask' given as CSS
** (inkscape-bin:1123): WARNING **: attribute 'mask' given as CSS
...
```
 But it does generate an image. My guess is these errors come from the fact that the mask is used via the style attribute in stead of using the mask attribute on line 146, 152 and 181. When I modify these lines to use the mask attribute, the error disappears.

 This leaves us with the real issue: The solder pads remain black (as opposed to a copper color) And the board soldermask remains square although the board's edge cuts is a circle.

 See the following example of my result:

[image]

 Version Information:
 - Kicad: 5.1
 - MacOS: 10.14.3
 - Inkscape 0.92.4

Unfortunately I'm not experienced with Python and Kicad enough to solve this issue on my own. I'f I do manage to figure it out, I'll make sure to send a PR.

gregdavill, 2019-03-24:
Hi Michael,

Thanks for testing out the scripts.
I've seen the inkscape warnings since moving to Ubuntu, but as you've seen they don't cause any issue. (Thanks for the tip about the style/mask attributes, I'll fix that in the scripts.)

I've not used MacOS for many years, I know that for these scripts to work they do need to be using the same python interpreter as KiCad, or be linked the KiCad directory to find the correct modules to load. I'm not sure on the correct way to add these universally to the scripts.

Down to the main issue. I hit a similar issue when I moved to KiCad 5.1. 
The script is supposed to re-colour the SVG elements depending on the layer they appear in the KiCad pcb. However it appears that the Copper layers are remaining black.

I think you mentioned you had on twitter, but can you confirm that you have tried the MacOSX modifications you've listed above on the KiCad 5_1 branch?
https://github.com/gregdavill/kicadScripts/compare/kicad_5_1

The code to handle arbitrary shaped boards is non-trivial so I haven't implemented that feature yet. The soldermask should however extend out to the edge.
[image]

MichMich, 2019-03-24:
Hi Greg,

I completely missed the fact that there was a 5.1 branch. This seem to have solved the issue for me. Thanks! Together with the earlier mentioned changes it works on MacOS.

I'll check if I can make a version of the script that runs on both Ubuntu as well as Mac. If I manage to do so, I'll send you a PR.

Thanks again for your help!

closed 2019-03-24 by MichMich

MichMich, 2020-09-10:
I think I eventually got it working, burins has been over a year since I used it. So I can't give you more info. Sorry.