gregdavill/kicadScripts #3: Oval pads on rotated footprint appear horizontal

UnguidedFreedom, 2019-04-21:
Hello,

First of all thank you so much for those scripts, they are amazing and the renders are gorgeous! 

I have a board with a USB connector at a 35° angle. It has two oval pads, but on the renders, the pads show up horizontally instead of being at an angle.

I just thought I'd let you know :)

[image]

gregdavill, 2019-04-21:
I noticed this a few weeks ago with one of my board designs. Except mine were only 0°/90°/180°/270° 

I've just pushed some code that may help. 
Can you test it and let me know, a quick test here on KiCad 5.1 with a 35° slot seems to work.

UnguidedFreedom, 2019-04-21:
Thank you for your fast reply! 

I have tested the updated code, and the hole is displayed at a different angle but it still isn't the correct one. 

If it helps, the pad is rotated 90° within the footprint, and the footprint is rotated 215° on the board.

[image]

gregdavill, 2019-04-21:
Thanks for testing that! I think my code is actually mirroring the rotation. (ie rotating in the opposite direction).

I have reproduced your results and just pushed an update.

UnguidedFreedom, 2019-04-21:
It works perfectly now, thank you very much!

closed 2019-04-21 by UnguidedFreedom