gregdavill/kicadScripts #12: No silk on rasterizing

joric, 2020-04-28:
No silk for resistors here: https://github.com/joric/nrfmicro/tree/devel
[image]
Maybe you could figure why faster than I.
Also wrong PCB shape but that's another issue.

gregdavill, 2020-04-28:
By default I don't think it renders references or values.

Try changing these lines here: https://github.com/gregdavill/kicadScripts/blob/master/plot_board.py#L417
```diff
- popt.SetPlotReference(False)
+ popt.SetPlotReference(True)
- popt.SetPlotValue(False)
+ popt.SetPlotValue(True)
```

joric, 2020-04-28:
Cool thanks it works now. What about cutting the board outline with edge cuts would that be a problem? It's just always rectangular now.

gregdavill, 2020-04-28:
Cool.

I'm not sure there is an easy way with the current method of rendering to cut out board shapes. I think We'd have to manually create a filled polygon that is filled, that would replace the background. But there are some tricky edge cases to handle:
- Multiple islands
- Internal cutouts defined by `edge.cuts`

If you have an idea on implementation I'd be happy to look it over and merge it.
Initially this script was really just designed to create textures that can be used in Blender/Fusion360, where they will clip out parts of the image that extend over the boards geometry.

Alright I'll close this.

commit 545a1da4 (2020-05-31): fixed #12 (enable rendering of references and values)
commit 85497088 (2020-05-31): Merge pull request #14 from joric/patch-1

fixed #12 (enable rendering of references and values)
closed 2020-04-28 by gregdavill