gregdavill/kicadScripts #16: Mask errors in Inkscape >1.0

gregdavill, 2020-06-16:
Soldermask generation no longer includes cutouts when rendering with inkscape >1.0.

I suspect this is to do with a different parsing method of the svg structure in the new inkscape, our svg mask is no longer being generated correctly.

korken89, 2020-06-30:
Hi I just stumbled onto this.
I hope it gets solved :)

mhollands, 2020-08-24:
Also running into this, but I have a workaround by manually modifying the SVG in Inkscape.

Firstly comment out the final line of the script that deletes the temporary file: `# shutil.rmtree(temp_dir, ignore_errors=True)`

Run the script and you will see a file called plot/temp/<my-board>-merged.svg

Open in Inkscape, right click on the solder mask shape and release the mask. This reveals the solder mask shape and the mask, but there appears to be an additional white rectangle that is breaking the mask. Delete this white rectangle. Select the solder mask shape and the correct mask. Object->Mask->Set Inverse. Then use Inkscape to export as PNG.

If I get time I might look into the fix and do a PR.

gregdavill, 2020-08-24:
I do have a fix for this, just need to push it.

gregdavill, 2020-10-30:
Fixed: https://github.com/gregdavill/kicadScripts/commit/0f579d20d8889ce78545c2e21d396c0f765b349b

closed 2020-10-30 by gregdavill