gregdavill/kicadScripts #17: KiCAD 5.1.6 not working in combination with plot_board.py

josephpal, 2020-06-26:
Okay, so far thanks a lot for this awesome script, regarding this [video](https://www.youtube.com/watch?v=L25B5WaZ2EI&t=544), it seems very nice and easy to use, but.....

It tried it out yesterday and it didn't work for me. My setup:
- KICAD 5.1.6 installed via ppa
- Linux Mint 19, Cinnamon
- Inkscape 0.92.3
- Python: Python 2.7.17, Python 3.6.9
`python --version` or `python3 --version`

So basically, when I run `python plot_board.py` with the input and output parameters, I got the following error messages:
`PCBNew not found, are you using KiCAD included Python ?` But in fact, I have installed KiCAD and regarding this command, I have python and the module `pcbnew` installed as well:

```
~ dpkg --listfiles kicad | grep python
/usr/lib/python3  
/usr/lib/python3/dist-packages 
/usr/lib/python3/dist-packages/_pcbnew.so 
/usr/lib/python3/dist-packages/pcbnew.py 
```

It seems to be that my KiCAD installation is now linked to the `python3` and not to `python version 2.7.17` as provided by the script. Maybe it is because KiCAD now switched to the newer python version... So i went further to get the job done by trying out running the script with `python3`:

```
~ python3 plot_board.py FILE.kicad_pcb
... 
Merging layers...
Rasterizing...
Traceback (most recent call last):
  File "plot_board.py", line 438, in <module>
    render(plot_plan, project_name + '-Front.png')
  File "plot_board.py", line 356, in render
    if len(version) > 1 and version[1] < "1.0":
TypeError: '<' not supported between instances of 'bytes' and 'str'
```

This issue could be fixed by removing the `and version[1] < "1.0"` in line 356. I am not sure if this is the way to fix it, but with this I was able to go ahead to went into the next issue :D:D:D:

```
~ python3 plot_board.py FILE.kicad_pcb
...
Merging layers...
Rasterizing...

** (inkscape:7896): WARNING **: 08:19:53.558: Cannot parse export area '71.89547122100001:323.425711221:955.161148779:736.3570487789999'; use 'x0:y0:x1:y1'. Nothing exported.
Traceback (most recent call last):
  File "plot_board.py", line 438, in <module>
    render(plot_plan, project_name + '-Front.png')
  File "plot_board.py", line 363, in render
    final_svg,
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
```

Okay, again an issue, but nothing to worry about. I thought maybe it is an "casting" issue and Inkscape doesn't accept float values to export an area from an svg file. So i tried it out by changing line 359 and line 368 from 

`format(x0,y0,x1,y1)` to `format(int(x0),int(y0),int(x1),int(y1))`.

Yeah and after that, I finally got the output I was searching for:

```
~ python3 plot_board.py FILE.kicad_pcb
...
Merging layers...
Rasterizing...
DPI: 1200
Background RRGGBBAA: 998060ff
Area -955:323:-71:736 exported to 11050 x 5163 pixels (1200 dpi)
Bitmap saved as: /DIRECTORY/FILE-Back.png
```

**Okay, to sum it up:**

1. First we have to check which python version your KiCAD version is working with. In case of `python3`, we have to run the script with the command `python3` instead of `python`
2. A few script changes which are mentioned above has to be done: 
   -  line 356: remove  `and version[1] < "1.0"`
   -  line 359 and line 368: change `format(x0,y0,x1,y1)` to `format(int(x0),int(y0),int(x1),int(y1))`

**Note:** This fix worked for me and my setup. I didn't tried it out on Windows or other Linux distributions. Moreover, I didn't try the `plot_gerber.py` script at all, so I have no hints what to change there to get it working. I am not quite sure if my provided solution is a good one, because I am not familiar with python at all, I am more a C++ / Java guy :D. 

What is left to do:
- I will create a fork of your script and add my changes 
- @gregdavill is it possible that you check my provided solution whether it makes sense or not?
- `plot_gerber.py` script has to be checked as well
- We have to take int account some other operating systems and setups to verify the solutions as a proper one

I hope I was able to help others who faced or will face the same problem ;) If any questiosn are left, feel free to contact me!

josephpal, 2020-06-26:
Fork now available under: [https://github.com/josephpal/kicadScripts](https://github.com/josephpal/kicadScripts)

korken89, 2020-06-30:
Hi

`version[1] < "1.0"` should be replaced with `float(version[1]) < 1.0`.
Then it will do the right thing :)

josephpal, 2020-06-30:
> Hi
> 
> `version[1] < "1.0"` should be replaced with `float(version[1]) < 1.0`.
> Then it will do the right thing :)

Ah nice, i will check it out ;)

gregdavill, 2020-10-30:
Is there still an issue running these scripts? 
I'll close this issue, but feel free to re-open or create a new issue if you encounter any issues.

closed 2020-10-30 by gregdavill