gregdavill/kicadScripts #18: Transparent areas turn black in Fusion360

korken89, 2020-06-30:
Hi, when having transparent areas i.e. no copper and no soldermask, the board turns black as there is no color there when using the rendered image as a material (not decal).
This could be fixed by generating a layer of yellow-ish to represent the PCB FR4 color and have it as the base layer.

gregdavill, 2020-07-08:
When rendering the final images, they shouldn't have any areas with an Alpha channel.

This is achieved by passing a background colour to inkscape, generally this colour is a brown/yellow colour that is supposed to resemble bare FR4 without copper or soldermask.

Maybe this is a bug in passing parameters into inkscape?

korken89, 2020-07-08:
Hi Greg!

It could be, when looking at the final "merged" SVG there is no background layer.
But the support for inkscape 1.0 has a few problems which can be the cause as well.

For info I run: `Arch Linux 5.4.50-1-lts`

gregdavill, 2020-07-09:
I have some local changes for 1.0 that I should push to a new branch.

Here is how I'm calling inkscape.
```
	subprocess.check_call([
		'inkscape',
		'--export-area-drawing',
		#'--export-area={}:{}:{}:{}'.format(int(x0),int(y0),int(x1),int(y1)),
		'--export-dpi={}'.format(dpi),
		'--export-filename', final_png,
		'--export-background={}'.format(colours['BackGround'][0]),
		'--export-background-opacity={}'.format(255),
		final_svg,
	])
```

I think the key here is the `--export-background-opacity={}'.format(255)`

gregdavill, 2020-09-15:
I've pushed this change to https://github.com/gregdavill/kicadScripts/commit/4aa7d49e8da7a8259c98f07b2178bf7ab778e2d7
Does this fix the issue?

closed 2020-10-30 by gregdavill