gregdavill/kicadScripts #19: Fail to find inkscape on windows

gregdavill, 2020-08-11:
Issue received via email:

> I just started using your software called "plot_board.py" and immediately had some errors that I don't know how to fix. I would open an issue in GitHub, but unfortunately I don't actually know what is failing. Here is the output I got:
> 
> ```
> [A bunch of numbers]
> 
> Merging layers...
> Rasterizing...
> Traceback (most recent call last):
>   File "plot_board.py", line 439, in <module>
>     render(plot_plan, project_name + '-Front.png')
>   File "plot_board.py", line 356, in render
>     version = subprocess.check_output(['inkscape', '--version'], stderr=subprocess.STDOUT).split()
>   File "c:\Program Files\KiCad\lib\python2.7/subprocess.py", line 217, in check_output
>     process = Popen(stdout=PIPE, *popenargs, **kwargs)
>   File "c:\Program Files\KiCad\lib\python2.7/subprocess.py", line 325, in __init__
>     errread, errwrite)
>   File "c:\Program Files\KiCad\lib\python2.7/subprocess.py", line 575, in _execute_child
>     startupinfo)
> WindowsError: [Error 2] The system cannot find the file specified
> ```
> 
> Do you know what's going on here? Is it a problem with Inkscape? I have Inkscape 0.92, which I installed using just the normal installer. I'm on Windows.
> Any help is appreciated!

This seems to be related to finding the path to inkscape. I'm generally using these scripts on linux, where calling `inkscape` just works. But I think on windows we'll need to actually define the full-path. 

If you want to try it, you'll need to edit these calls to inkscape: https://github.com/gregdavill/kicadScripts/blob/master/plot_board.py#L356-L375

jacobtohahn, 2020-08-11:
changing "inkscape" to the path to inkscape does fix the issue. The front of the board does rasterize. However, when rasterizing the back of the board, another error is thrown:

```Merging layers...
Rasterizing...
Traceback (most recent call last):
  File "plot_board.py", line 452, in <module>
    render(plot_plan, project_name + '-Back.png')
  File "plot_board.py", line 374, in render
    final_svg,
  File "C:\Program Files\KiCad\lib\python2.7/subprocess.py", line 191, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Program Files\\Inkscape\\inkscape', '--export-area=-239.872021221:169.847151221:-719.993978779:553.982508779', '--export-dpi=1200', '--export-type=png', '--export-filename=C:\\Users\\jtoha\\OneDrive\\Documents\\KiCad Projects\\ZTO-80\\SBC\\plot\\SBC-Back.png', '--export-background', '#998060', 'C:\\Users\\jtoha\\OneDrive\\Documents\\KiCad Projects\\ZTO-80\\SBC\\plot\\temp\\SBC-merged.svg']' returned non-zero exit status 1```

Any fix for this?

gregdavill, 2020-08-12:
Can you try running this in a command prompt. Maybe the error message it outputs will be more useful?
```
C:\Program\ Files\Inkscape\inkscape --export-area=-240:170:-720:554 --export-dpi=1200 --export-type=png --export-filename=C:\Users\jtoha\OneDrive\Documents\KiCad\ Projects\ZTO-80\SBC\plot\SBC-Back.png --export-background #998060 C:\Users\jtoha\OneDrive\Documents\KiCad\ Projects\ZTO-80\SBC\plot\temp\SBC-merged.svg
```

jacobtohahn, 2020-08-12:
It did give a better output. Here it is:

```
** (inkscape.exe:46432): WARNING **: 20:16:04.613: Invalid option --export-type=png
```

Is that any different from the front side, which did rasterize?

gregdavill, 2020-08-12:
Weird, it shouldn't be any different. 

Try using `--export-png` instead. I think that is one of the options that has changed in Inkscape 1.0+

jacobtohahn, 2020-08-12:
Tried that. After running:
```
"C:\Program Files\Inkscape\inkscape" --export-area=-240:170:-720:554 --export-dpi=1200 --export-png --export-filename="C:\Users\jtoha\OneDrive\Documents\KiCad Projects\ZTO-80\SBC\plot\SBC-Back.png" --export-background #998060 "C:\Users\jtoha\OneDrive\Documents\KiCad Projects\ZTO-80\SBC\plot\temp\SBC-merged.svg"
DPI: 1200
```
A new error appears:
```
** (inkscape.exe:31840): WARNING **: 20:23:08.058: File path "--export-filename=C:\Users\jtoha\OneDrive\Documents\KiCad Projects\ZTO-80\SBC\plot\SBC-Back.png" includes directory that doesn't exist.
```

I can confirm that this directory does, in fact, exist.

gregdavill, 2020-08-12:
I'm not entirely sure what's going on now :/

I wonder if this is related to the spaces in the path?

jacobtohahn, 2020-08-12:
Perhaps, but adding quotes to the file paths in python or just in the command line doesn't fix it. Is it possible to skip the rasterization step so I can do it manually?

jacobtohahn, 2020-08-12:
FIXED:
I replaced:
```
else:
		subprocess.check_call([
			'inkscape',
			'--export-area={}:{}:{}:{}'.format(x0,y0,x1,y1),
			'--export-dpi={}'.format(dpi),
			'--export-type=png',
			'--export-filename={}'.format(final_png),
			'--export-background', colours['BackGround'][0],
			final_svg,
		])
```
with:
```
else:
		subprocess.check_call([
			'C:\Program Files\Inkscape\inkscape',
			'--export-area={}:{}:{}:{}'.format(x0,y0,x1,y1),
			'--export-dpi={}'.format(dpi),
			'--export-png', final_png,
			'--export-background', colours['BackGround'][0],
			final_svg,
		])
```
Which is the same code as is in the main part of the if statement. It may have not been obeying the if statement or something? You know the code far better than I do; I'll let you decide what happened.

gregdavill, 2020-08-13:
The if statement is supposed to determine if inkscape is version <1.0 or 1.0+ because they have different requirements on parameters. 

Looks like this whole section should be re-worked.

Thanks for poking around at this, glad that it's in a working state for you now.

gregdavill, 2020-09-15:
I think some recent changes have fixed this. 
Looks like the version check might have been failing.

jacobtohahn, 2020-09-15:
Ok, awesome. I’ll test it out later.

jacobtohahn, 2020-09-15:
Unfortunately, this still doesn't work, at least for now. You seem to be using F strings, which are supported in Python 3, but not Python 2. The version of Python used by KiCad 5 (at least on Windows) is Python 2, and thus this script will not work.
However, KiCad 6 should be migrating to Python 3 (or so I've heard), so this script should work in that version.

gregdavill, 2020-09-15:
Whoops, Sorry. 
Python 2 was deprecated at the start of the year, but I'd forgot that it was still packaged with KiCad 5.
I removed that f-string, for a `format()`

jacobtohahn, 2020-09-15:
Luckily, removing the f-strings allowed the program to run, but I still recieve an error:
```
C:\Users\jtoha\OneDrive\Documents\KiCad Projects\Adder>"C:\Program Files\KiCad\bin\python.exe" "C:\Users\jtoha\OneDrive\PCB\kicadScripts\plot_board.py" ".\*.kicad_pcb"
Merging layers...
Rasterizing...
Detected Inkscape version 1.0+
Command '['inkscape', '--export-area-drawing', '--export-dpi=1200', '--export-type=png', '--export-filename=C:\\Users\\jtoha\\OneDrive\\Documents\\KiCad Projects\\Adder\\plot\\_autosave-Adder-Front.png', '--export-background', '#998060', '--export-background-opacity=255', 'C:\\Users\\jtoha\\OneDrive\\Documents\\KiCad Projects\\Adder\\plot\\temp\\_autosave-Adder-merged.svg']' returned non-zero exit status 1
Merging layers...
Rasterizing...
Detected Inkscape version 1.0+
Command '['inkscape', '--export-area-drawing', '--export-dpi=1200', '--export-type=png', '--export-filename=C:\\Users\\jtoha\\OneDrive\\Documents\\KiCad Projects\\Adder\\plot\\_autosave-Adder-Back.png', '--export-background', '#998060', '--export-background-opacity=255', 'C:\\Users\\jtoha\\OneDrive\\Documents\\KiCad Projects\\Adder\\plot\\temp\\_autosave-Adder-merged.svg']' returned non-zero exit status 1
```
The issue seems to be that it is detecting the wrong Inkscape version. I'm using version 0.92, not 1.0+.

gregdavill, 2020-09-16:
Hmm, interesting. Can you try adding a `print(version)` after the version check on line 365

```python
		version = subprocess.check_output(['inkscape', '--version'], stderr=None).split()
		print(version)
```

jacobtohahn, 2020-09-16:
Well, interestingly, the version output is... nothing!
```print(version)``` returns [].
I also checked this by typing ```inkscape --version``` in a command prompt and got the same thing, just a blank line.
I'm sure this is a bug with Inkscape. Maybe it's just my system. I'm also not sure if this occurs in Inkscape 1.0.

gregdavill, 2020-09-16:
Hmm, weird. Well I suppose we can assume version < 1.0 if the fields are blank.
Just pushed another change

gregdavill, 2020-10-30:
I will assume this is now fixed. Feel free to re-open if that is not the case.

closed 2020-10-30 by gregdavill