gregdavill/kicadScripts #20: No Attribute 'SetLineWidth' in KiCad nightly

Tobi4sDE, 2020-10-26:
Hi,

I'm using a pretty recent nightly version of KiCad and always get this output when using your scripts: 

`C:\Users\Tobias\Desktop\kicadScripts-master>"C:\Program Files\KiCad\bin\python.exe" plot_board.py "C:\Users\Tobias\Documents\GitHub\uDice\uDice\uDice.kicad_pcb"
Traceback (most recent call last):
  File "plot_board.py", line 425, in <module>
    popt.SetLineWidth(FromMM(0.35))
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetLineWidth'`

Not sure if it's just the attribute name that has been changed or whether there are more changes.

gregdavill, 2020-10-30:
I actually don't think that call is required when plotting with SVG, it appears to be missing in v6. Which is kind of handy because we need a good way to detect if we are using KiCad v5 or KiCad v6, because there are some other changes needed to support the output from v6.

Can you give this version a try: 0f579d20d8889ce78545c2e21d396c0f765b349b

Tobi4sDE, 2020-10-30:
Thanks a lot for looking into the issue!

Running the script doesn't show any errors any more:

```
C:\Users\Tobias\Desktop\kicadScripts-master>"C:\Program Files\KiCad\bin\python.exe" plot_board.py C:\Users\Tobias\Desktop\uDice\uDice.kicad_pcb
Merging layers...
Rasterizing...
Detected Inkscape version 1.0+
Merging layers...
Rasterizing...
Detected Inkscape version 1.0+
```
But this is all I'm getting, just the front, no back side file gets created. I also tried it with a few different board files, with the same result.

[image]

I upgraded to the latest nightly build and Inkscape is v1.0.1

gregdavill, 2020-10-30:
Hmm. Are you able to share your KiCad design? I'll run some tests on windows tomorrow.

Tobi4sDE, 2020-10-30:
Sure, here are the two boards I was trying it with: 
[KiCad_files.zip](https://github.com/gregdavill/kicadScripts/files/5466118/KiCad_files.zip)

gregdavill, 2020-10-31:
Strange, it appears the scaling of layers is different from the SVGs generated on a windows machine.

I don't have much more time today to debug unfortunately.