gregdavill/kicadScripts #24: Broken invert function when drawing the solder mask

cchaine, 2021-11-28:
I am running nightly release `20211127-055350-b8b8c0d34` on macOS and I've experienced issues when plotting a board, where the solder mask wouldn't be drawn.

I have investigated a little and it happens within the `addSvgImageInvert` function, where we would like to create a rectangle the size of the board and use the top layer as a mask on that rectangle.

It appears that the `GetWidth`, `GetHeight` functions both return 0, which explains why there isn't any solder mask on the outputted image.

I haven't found anyone else having this issue unfortunately. Did you ever come across anything similar ?

[image]

cchaine, 2021-11-28:
I don't know if this issue is related to my board, to the nightly release or my environment.

I managed to find a workaround by manually drawing a rectangle to find the width, height and origin values for the board.

[image]

gregdavill, 2021-11-29:
Thanks for debugging this. I've actually not used these scripts in awhile, so likely it's some changes needed for supporting nightly rather than your design or environment.

cchaine, 2021-11-29:
No worries. They either changed the api and haven't updated the documentation or this feature is just broken in the nightly build.
I'll just close this issue and submit one to the kicad repository.

closed 2021-11-29 by cchaine